rhapsodyv/Marlin #10: Autoleveling button do nothing

eclips16, 2020-11-09:
I use
#define V6_330_TITAN_TMC 1
#define TFT_LVGL_UI

I am trying to run the autolevevel by using the button in the tools menu bug nothing happen even after configuring the gcode command for autolevel (G28;G29)

blayno, 2020-12-26:
yes i have this same problem also, i wonder if there will be a fix soon

dandantsui, 2021-01-02:
perhaps if @rhapsodyv merged from main Marlin branch -- it would be fixed?

kondorzs, 2021-01-03:
Hi I have Auto level button problem, but I upgraded more and more and sometimes work. I have probelm Cura MKS plugin (prewiew) fno work in file browser just work and Show image on printing screen.
I ame transolate Hungarian (TFT) language and I would like to change Traditional Chinese to Hungarian.
I tested mor and look for problem Autolevel and Prewiew.

PY2BND, 2021-01-09:
Yes, I had the same finding, ABL under TFT_LVGL_UI isn't reacting on the screen.

Pigi-102, 2021-03-27:
I've the same issue, but have a finding too.
I have serial connected and when I push the ABL button from UI, in the serial output i get:
`ok
echo:Unknown command: "ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ"
ok
`
So it's probably some assignment wrong in the routine

m0uh85, 2021-04-23:
is the problem fixed? because I prefer the LVGL UI instead the COLOR UI but without ABL makes no sense to compilie with this UI

Pedro8005x, 2021-04-23:
Hi,
When I selected AUTO_BED_LEVELIG_BILINEAR, there is manual bed level icons
for corners under abl button, but mesh is unavailable for me if LVGL is
selected. If thete is any news on this topic, I'm interested too...

m0uh85 ***@***.***> ezt írta (időpont: 2021. ápr. 23., Pén
12:59):

m0uh85, 2021-05-03:
I compiled last week with LVGL and used the ABL many times without any problems - seems to be fixed!

rossosat, 2022-02-01:
> 

Can u help me? i have the same problem

Pedro8005x, 2022-02-01:
Hi. Try to make an auto home and then push auto bed level button. No any
GUI response, but worked for me. All in all, a flat bed is the best with
manual adjustments, i switched off abl completely...

rossosat ***@***.***> ezt írta (időpont: 2022. febr. 1., K
20:04):