rhapsodyv/Marlin #14: CXY-V6, linear advance

codemunzlp, 2020-11-29:
The Linear Advance setting causes the Extruder Motor of my X5SA-400 with the CXY-V6 Board to stop mid print. It won't move until I turn the power off and back on. There is no clog, the E motor just stops working.
[Configuration &  Configuration_adv.zip](https://github.com/rhapsodyv/Marlin/files/5612507/Configuration.Configuration_adv.zip)

codemunzlp, 2020-11-29:
1st Step: activate linear advance in marlin firmware
2nd Step: Print something for a few hours, the E motor will stop suddently

codemunzlp, 2020-11-29:
It should be fixable using this:
[image]

rhapsodyv, 2020-12-24:
Where is this image from?

dandantsui, 2021-01-02:
Looks like it is from here

https://www.thingiverse.com/thing:4588538

Eliminateur, 2021-04-11:
Indeed, you can't use linear advance with the steppers in stealthchop mode, that's set from factory, linear advance requires spreadcycle mode.
That image you pasted is not exactly the best, as that would change the mode permanently but you'll remain in standalone mode. 

for the best operation they need to be in UART mode so that marlin can change between modes as it wants(plus a lot of other advanced stuff, like running the extruder in spreadcycle alone for the torque)

this caveat(about the modes) should be added as a comment block on the file before linear advance, as right now it says nothing.

corrado-c, 2022-01-23:
some time ago I made the change to activate the UART, but it didn't work for me,
Are there any news or better instructions?
[image]