rhapsodyv/Marlin #15: Filament Sensor always on

Surrealcrow, 2020-12-10:
### Description

Printer: Tronxy XY Pro 2, no mods, no titan extruder.

Migrated from stock firmware to Marlin, after printing some parts realized that the filament sensor is not working, physically it turns off the light when filament runs over, double checked that it is activated on screen and with gcode and it always shows as triggered.

I downloaded the code and uncommented the lines for my printer model

### Steps to Reproduce

1. Turn the sensor on 
2. remove the filament 
3. start a print
4. starts printing

1. Turn the sensor on
2. execute M119 with filament 
3. M119 response filament:  TRIGGERED
4. Remove filament and execute M119
5. M119 response filament:  TRIGGERED

If i get back to the original stock firmware, the runout filament works

Will add the conf.h files when get home
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

toxic-garden, 2020-12-24:
Here's a fix for that
https://github.com/rhapsodyv/Marlin/compare/tronxy-full-mks-tft...toxic-garden:fix_x5sapro_filemant_runout?fbclid=IwAR3_Mm2UwDK1W1Gdugq6grKjh-KM_SUpR4uJJDnguAg2R_wvgThm0OidIMw

toxic-garden, 2020-12-24:
and yes, this is a fix for the X5SA Pro, but i heavily assume that the XY Pro 2 uses the same sensor technique, so it might be easily adaptable for XY Pro 2

rhapsodyv, 2020-12-24:
Fix Merged

closed 2020-12-24 by rhapsodyv