rhapsodyv/Marlin #19: Filament Runout Not Working

chucky3456, 2021-01-30:
CXY-V6-191017

Filament Runout Sensor not pausing print when filament runs out (not a critical issue).

Steps: 1. Start print with limited filament
2. Let filament run out
3. Print continues

Eliminateur, 2021-04-11:
¿Have you checked the sensor state with M119 first?

m0uh85, 2021-05-03:
Checked M119 Sensor states are OK but same issue for me the printer is still working after run out of filament.

Eliminateur, 2021-05-03:
¿are we talking printing from the SD card or from octoprint/host via USB?

m0uh85, 2021-05-03:
Octoprint... I saw something in the config:

 * IMPORTANT: Runout will only trigger if Marlin is aware that a print job is running.
 * Marlin knows a print job is running when:
 *  1. Running a print job from media started with M24.
 *  2. The Print Job Timer has been started with M75.
 *  3. The heaters were turned on and PRINTJOB_TIMER_AUTOSTART is enabled.

is this the reason when I start printing from Octoprint? Marlin didn't get one of these informations?

Eliminateur, 2021-05-03:
octoprint does not support filament runout sensors connected to the board itself, as apparently there's no gcode or response than marlin can provide to octoprint that the runout sensor has stopped.
From what i researched when i wanted to make this work as well, you need to connect the runout sensor to the rpi GPIO ports, and run the runout detection on the rpi itself with a octoprint plugin.

I find that a kludge at best and very daft, and i don't plan on doing it in the least, specially since there's no clean way to do it on tronxy printers with the ribbon cable going straight to the board.

m0uh85, 2021-05-04:
> octoprint does not support filament runout sensors connected to the board itself, as apparently there's no gcode or response than marlin can provide to octoprint that the runout sensor has stopped.
> From what i researched when i wanted to make this work as well, you need to connect the runout sensor to the rpi GPIO ports, and run the runout detection on the rpi itself with a octoprint plugin.
> 
> I find that a kludge at best and very daft, and i don't plan on doing it in the least, specially since there's no clean way to do it on tronxy printers with the ribbon cable going straight to the board.
> 
> Edit: i've found this https://community.octoprint.org/t/filament-runout-sensor-connection-rpi-gpio-versus-printer-board/29725
> that apparently enabling host action commands in marlin would make marlin send a command about the runout, i haven't tried it yet

Make senses... Thanks!