rhapsodyv/Marlin #20: Tronxy XY 2 PRO TFT_LVGL_UI not booting

kurtisane, 2021-02-03:
Hi,
I build the project with 
#define XY2_V6_255_TITAN_TMC 1
#define TFT_LVGL_UI

This does not boot up. It seems to flash correct (couple beeps while flashing). 
But then it powercycles.

Any ideas why ?

nickgal, 2021-02-04:
Did you copy the assets folder to the SD card as well?

https://github.com/rhapsodyv/Marlin/blob/38ea7b62f9b38a61364b26b59fa78009f6e4f014/Marlin/Configuration.h#L2854-L2855

Etienne28051968, 2021-03-01:
Currently running into the same issue, also on 

#define XY2_V6_255_TITAN_TMC 1
#define TFT_LVGL_UI

And yes the assets directory from the build is on the SD card, when I power on my Tronxy, I hear two beeps, then several seconds later again two beeps, but further nothin anymore, also screen stays black.

Is it correct that only assets directory and the update.cmb need to be on the SD card ?

Thanks in advance...

Etienne

richardmaltais, 2021-03-06:
@kurtisane I had the same issue but after reading this comment from @nickgal , realized I didn't. Pretty sure it'll solve the issue for you as well.