rhapsodyv/Marlin #26: Documentation

thawkins, 2021-03-14:
It should be documented that if you are building for a Chitu V6 board that the following line in platformio.ini  (near top of the file)

"default_envs = chitu_v5_gpio_init # LPC1769"

should be changed to 

"default_envs = chitu_f103" 

so that the binaries are built for the STM32F1 when using the default "clean all" and "build all" actions in VSCode

orangeworx, 2021-03-22:
You need not change anything in platform.io.
I've successfully compiled and flashed a v6 by following the example files & readme
The change is done in Configuration.h around line 135
` // Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_CHITU3D_V6
#endif `
VS code gets the dependencies at build time

Eliminateur, 2021-04-11:
There's no need for that, you can use chitu_v5_gpio_init for the v6 board as well(tested it).
If you check line 1154:
```
platform      = ${common_stm32f1.platform}
extends       = env:chitu_f103
build_flags   = ${env:chitu_f103.build_flags} -DCHITU_V5_Z_MIN_BUGFIX
```
It uses the f103 settings and adding the z min bugfix

orangeworx, 2021-04-13:
> There's no need for that, you can use chitu_v5_gpio_init for the v6 board as well(tested it).
> If you check line 1154:
>
> ```
> platform      = ${common_stm32f1.platform}
> extends       = env:chitu_f103 [...]

there has been no need to modify the platform.io file in the most recent building process... everything can (and should, from my understanding) be done in configuration.h and configuration_adv.h

Eliminateur, 2021-04-13:
i haven't tested that, ¿so it doesn't matter what you set your default_envs?(or you have to comment the line?), because in my case, changing it(or not) in in the ini can break Z-limit, if i don't specifically set the correct environment. Example:
board declared as chitu_v6 in configuration.h, board set as "default_envs = chitu_f103" in platformio -> Z endstop fails.
now board defined the same but environment changed to chitu_v5_gpio_init -->works correctly.

So there is in fact a need to modify the .INI

orangeworx, 2021-04-21:
TBH, I cannot confirm this as I've not had to edit the platformio.ini and have not had issues with VSCode getting the dependencies for the v6 board I'm running. I do know that it's now recommended to use VSCode Vs Arduino IDE, with the Auto Build Marlin addon, though for the Chitu boards, you'll need to grab the cbd fw file and SD flash to board instead of flashing directly from within Auto Build Marlin.

HTH