rhapsodyv/Marlin #27: tronxy x5sa pro no usb connection

secretman28, 2021-03-14:
i got to compile and install the firmware in the chitu mb,  all work inboard, but cant get any program to connect to it , did i need to change something ine the config ?

orangeworx, 2021-03-23:
Do you see a COM port usage when you connect the printer via USB?

secretman28, 2021-03-23:
yes it show a com port, but when I'm removing the ad card it disapear it's weird all the rest work perfect LCD  print etc I manage to get the uart 3 link to the wifi port it work but too buggy printer take pause even with a esp-02 really Wana fix the usb but it's over my competence. but it's suppose to work ?

orangeworx, 2021-03-23:
If I understood correctly, you have an SD in the slot and when you remove the SD card (Attach/Remove media or just pull SD out?), the usb connection drops? As in, you see the COM port being released in device manager (I'm assuming you're on Windows).
That is strange behavior, I'll attempt to recreate this as soon as I have access to my printer.

I'll get back to you on that and I'll include my config files for you to compare.
cheers

secretman28, 2021-03-23:
yes but I never get connection to the printer at any time I can connect to the port but nothing I will include my config file tonight home

secretman28, 2021-03-25:
[Marlinx5sa config.zip](https://github.com/rhapsodyv/Marlin/files/6201120/Marlinx5sa.config.zip)
 here is the config all work like a charm except cant connect usb, the uart is change to accept the wifi esp-02 for now that work soso wifi connection or the code i did is not the best haha

secretman28, 2021-03-27:
any idea ?

orangeworx, 2021-03-29:
Hello jp, sorry about the delay just been caught up...
I did a side-by-side file compare (i use winmerge) and you seem to have many differences...
The motherboard definition is certainly wacky... completely different in my configuration.h... A couple things I believe you need to check are the baudrate, it's 250000 on the tronxy and you also need to uncomment #define COREXY
uncomment #DEFAULT_AXIS_STEPS_PER_UNIT  (for the titan extruder)
you have 2 nozzle to probe offsets (if you are using the stock extrusion head/probe, use my values, I measured them, confirming example config)
uncomment X_BED_SIZE and Y_BED_SIZE
 missing #define Z_MAX_POS 400
comment FIL_RUNOUT_PULLDOWN
I'm using bilinear bed leveling, yours is set for UBL (just validate it's what you want)
change GRID_MAX_POINTS_X to 4 (to have 16 point bed leveling, like original)
comment EXTRAPOLATE_BEYOND_GRID
uncheck your TFT interface FSMC
and uncheck TFT_RES_480x320
I have this TFT_LVGL_UI unchecked as I'm trying to compile my marlin 2.0 with this... it's failing for me but seems some have been successful.

As for your USB connection, first thing I'd try is to set the baud rate to 250000
Where did you grab this config file from?

secretman28, 2021-03-29:
i just start from the downloaded git and mod my option the lcd was neededto be change the s-curv ubl its me that i put it i will try to put the 250000 but on the esp-02 its hard to make it happen but if work on usb will be perfect no need anymore 
i will try your config this week and report i dont know why it is different... before vscode compile on my pc i took a lot of debug of the vscode not the marlin... can you share the update file to put in the sdcard i will try your compile here and your compile by you , in case its my computer that is shitty lol

orangeworx, 2021-03-29:
Did you grab the configuration file examples from the repo?
once you have marlin extracted and folder added to VSCode, you can compile just to make sure everything is working with defaults.
You then copy configuration.h and configuration_adv.h to the marlin folder and adjust to your liking...
I have not fully validated that my config is ok... I know it's working and have printed successfully, bed leveling without issues.
My concern at the moment is relating to 2 issues
- the new TFT UI (which I still can't compile)
- homing/bed leveling: homing sends XY somewhere near the bed center but still pretty off and level probing points aren't exactly as stock, the 16 points aren't evenly placed.

secretman28, 2021-03-29:
yes I taken the config for start I think cuz I just change my normal setting and didint toutch the rest

secretman28, 2021-04-03:
ok i try all posible serial impossible to connect to the printer via usb
still do the thing with the sd card i put it computer detect something, pouu douuu douuu loll but nothing change in the pc in or out sdcard cannot connect im about to change the board :( i love the board work realy nice but i want usb connection 
i will try te redo a build for my esp and get better connection if you see something that i forget on the usb connection tell me 
i continu to test this weekend

orangeworx, 2021-04-03:
I'm curious about something...
When you initially worked on your motherboard, you removed the usb & 5v jumpers... did you replace these?
Aside from this, I'd try going back to chitu firmware (I contacted TronXY to send me their latest .cbd).
I had the original firmware saved before flashing marlin, I could include it here if you need it.

secretman28, 2021-04-03:
yes , jumper at the right place , even the stock firmware you can't control the printer by usb that's all the point at 1 I change it  , I got my own firmware aside in case was printing good but why have a machine that you can't modify hahaha still working on a solution

secretman28, 2021-04-07:
can you make me the update file x5sa pro like yours i dont care i wana try if its me or not i try anything cant get it to work
blue level sensor runout sensor bone stock lcd is color one and board v6 silent driver

orangeworx, 2021-04-07:
I'm gonna attach my latest, though I haven't had a chance to use it... There are slight differences between what I'm running right now (I'm printing with it atm)
This is still needing some adjustments... unfortunately I have had enough time to make sure all the options for this printer are on/off. I noticed a couple of things to look into.

[Config files](https://github.com/rhapsodyv/Marlin/files/6273017/Marlin.zip)

Let me know how it goes.

secretman28, 2021-04-11:
will try this week end sry for the delay

secretman28, 2021-04-12:
sorry mate but im out, i try realy hard but usb not work at all im out i just finish install a duet2wifi hope  you figure it out

orangeworx, 2021-04-12:
I've got most of my problems solved for now and it's working good... is the duet2wifi really worth it? I connect my printers to raspberrypies and leave them on, I always have access to them and it's been working awesome to be honest.

Let me know what you plan to do with the tronxy board.

secretman28, 2021-04-13:
ya till now duet is the king I got a maestro on my old x5s and it was the best upgrade I did it's a game changer smoother, more precise stable etc... I got the x5sa pro as 4 printer the x5s is my favorite of all the one I try  and really love the upgrade they made to it for the x5sa pro version the Chiru impress me at start silent seam precise too but limited mod all close source etc.. I got hope when I put marlin to it but no usb or wifi is a big no for me hate SDcard haha but except that marlin was working #1 no buy except for auto level after all it always try to print 25mm hi to the bed even changing all offset I can haha , I try mks gen2, mks genL, ramp, Chitu 1 version , Chitu last version, duet2 maestro,duet2 wifi, melsi ,and a bunch of them and once you get the new reprap way of showing thing,setting things, and even the wifi, or ethernet version can plus usb like a normal board to simplify3d,repetier host,matter control, etc .. the clone of the wifi work the same  I got one for my alfaweise u20 for 2 yrs and still run like a charm let's say it's 1/2 price and you should try hahahah

but still your doing a amazing work on this marlin adaptation, from the first try all work screen print #1 SDcard work no big bug keep your good work 
you deserve a 🍺  
I will still try to work it out but I Wana use the beast  a bit hahaha

secretman28, 2021-04-13:
for now I keep the board in the spare part  box, probably try to make this work later on ,  why you need a spare? lol

orangeworx, 2021-04-13:
> for now I keep the board in the spare part box, probably try to make this work later on , why you need a spare? lol

I thought if you were gonna get rid of it, I wouldn't mind picking it up from you and fiddling with it... I don't actually need a spare, I have an SKR 1.4 Turbo and a Creality 4.2.7 sitting in their boxes waiting to be opened! haha

secretman28, 2021-04-13:
haha ok when I decide what I do with it I will tell you and if the  I will ship you the thing

orangeworx, 2021-04-14:
that's nice of you, I'd pay the shipping and I'm willing to give you something for the board.
Glad you're happy with your current setup :D