rhapsodyv/Marlin #28: Extruder stops working near the end (Linear Adv is off)

willmackey2nd, 2021-03-20:
### Description

Extruder motor stops near the end of the print when there's fine details

### Steps to Reproduce

Noticed it occurring randomly (but often) with Marvin key chain model and then also with the Jungle boat. Boat takes ~3h to print with these settings so not a fun thing to try to reproduce. 

**Expected behavior:** Finish printing.

**Actual behavior:** Extruder motor stops and stays torque off for the rest of the print

#### Additional Information

I'm having an issue with the extruder motor stopping near the end of the print when there are fine details. This seems to only happen near the end of the print, not in the middle even with models with a lot of fine details (like the jungle boat).

I noticed this first with the small Marvin key chain, as sometimes the chain loop wasn't printing. Sometimes it was. I thought it was temperature issue. Then I tried printing this Jungle boat and noticed extruder stopping near the end when printing the chimney. Tried to print it again and it stopped pretty much exactly at the same spot (measured with caliber). If I print this chimney alone, (by dropping the model height in S3D), it prints fine, so nothing in the model as such.

Setup:
- Tronxy x5sa pro (bowden titan extruder clone), original V6 board.
- Mosquito Magnum clone hotend
- Marlin Tronxy fork https://github.com/rhapsodyv/Marlin/tree/tronxy-full-mks-tft/Marlin
- S3D slicer (0.15 layer height, 0.6mm nozzle)

This is not:
- Clogging. The motor just stops even trying.
- Linear advance issue as it is turned off (verified by M900 not being recognized)
- Stepper motor over heating. It's around 40C.
- Stepper driver over heating. I added a heat sink and A LOT of airflow. Haven't touched in the current adjustment so it's factory setting and as the motor isn't overheating, I doubt it's too much.

P.S. never mind the string and otherwise bad quality. Experimenting with this new hotend (and not succeeding)
[image]

[conf.zip](https://github.com/rhapsodyv/Marlin/files/6175957/conf.zip)

willmackey2nd, 2021-03-20:
Hmh, found out during some string tower tests prints that it seems to stop also earlier, not not just at the stop. I also heard a screech / messed up step sound before it stopped. I've checked the cabled, wobbled and tried to yank them during continuous extrusion test and can't find any connection faults.
Are these TMCs prone to malfunction this way? The machine hasn't been used but for maybe one whole roll worth.

willmackey2nd, 2021-03-20:
Also after failing and after the print has finished or the job has been cancelled, the extruder motor does work when trying to manually extrude from e.g. octoprint.

Pedro8005x, 2021-03-20:
I had the same problem with chitu (xy2pro) fw on x axis. I tried to rise
motor currents, additional cooling for motherboard, a lot of spent time to
tries. I bought another motherboard, no success, and then I bought a new
motor. The problem remains, in fact more rarely, but not good at all. I've
got some fw trial from Tronxy's customer service, but no success.Then I
began use Marlin. (Better but fails again at fast accelerations) The final
solution for me was: limit all speeds to 150mm/s, acceleration to 1500mm/s2
(x/y), after that I never seen motor stop again. It is a half-solution, but
no more hope remained to find better. This is not the only annoying issue
with this coffe-grinder, so I'm happy with this limitations... In fact,
extruder was good for me

willmackey2nd ***@***.***> ezt írta (időpont: 2021. márc.
20., Szo 19:36):

willmackey2nd, 2021-03-20:
Thanks @Pedro8005x for commenting and sharing.

Sorry for being a bit spammy but I opened an issue at the main marlin repo too as I wasn't sure if this one is a too active.
I went to really try to reproduce this and didn't take long to find a good model. The basic string test. That I managed to print once fully, but then 3 following tries made extruder stop _exactly_ at the same layer. This made me doubt it would be hw issue and as I noticed this screeching sound (usually) before the failure I doubt it's the slicer either.

Here's more about the new finding:
https://github.com/MarlinFirmware/Marlin/issues/21390

willmackey2nd, 2021-03-29:
@Pedro8005x, it could be that this indeed is either TMC driver issue or something as I too experienced stop with one of the XY motors too. I want to utilize full speed of this large scale printer so limiting speeds is really not ideal. It's not really a problem if I print 0.6x0.5 layers as even a Mosquito clone starts to bottleneck with those thicknesses and 100mm/s is enough there, but with thinner layers it just has to fly or everything gets painfully slow. For a hammer solution I ordered a Duet 2 Wifi clone for 60 bucks.

Closing this issue as a duplicate to the linked Marlin ticket. Hopefully @rhapsodyv or someone can give a better closure to this at some point in case it's a generic issue with these TMC2225 drivers or something.

closed 2021-03-29 by willmackey2nd