rhapsodyv/Marlin #37: CXY-V9-200415 Board

NickiAtNight, 2021-06-16:
Is support for this board here? If not, I can provide pictures. The stock firmware keeps crashing the bed into the nozzle so the printer is currently unusable. ANY help is appreciated.
[image]
[image]
[image]
[image]

NickiAtNight, 2021-06-16:
This is for the X5SA-500-2E printer with dual titan extruders and a 500x500 bed

greenaar, 2021-06-16:
I had one in hand recently, it's 2209 standalone drivers on Z1, z2, x and y, and 2208 standalone on E.  it's otherwise functionally a v6 board.

NickiAtNight, 2021-06-16:
So is there a way to configure one of these premade configs to work with it?

greenaar, 2021-06-16:
Take the pro 330 v6 config as a start.  switch the XYZ to 2209 standalone, add a z2 stepper also 2209.  Set the bed size.  

Look in configuration_adv.h for

#define NUM_Z_STEPPER_DRIVERS 1 (set it to 2)
Look at 
//#define Z_STEPPER_AUTO_ALIGN  (uncomment this around like 820)

Configure the z stepper auto align section (basically where to probe for bed level (that'll be at something like 50,250 and 450,250 on a 500mm bed.

VTXtruder, 2021-07-19:
A bit late on this but I have the same board on a X5SA-500-2E. It's close to the same board as a V6 but there are some pin changes. I have submitted pull requests to Marlin with the code to support it. So, hopefully mainline Marlin will have it soon. If you don't want to wait grab it from my repo.

https://github.com/VTXtruder/Marlin
https://github.com/VTXtruder/Configurations