rhapsodyv/Marlin #39: Encoder with Chitu

snovotill, 2021-07-17:
The Chitu CXY-V6-191017 actually supports an encoder for navigating the UI. The signals are present in the ribbon cable to the display as follows but they are not terminated on the display board. Note that Pin1 is transposed on the PCB so I've listed the conjugate pin numbers in brackets:

EXT_ENC_A | MCU Pin44 PC4 | Ribbon Pin2(39)
EXT_ENC_B | MCU Pin45 PC5 | Ribbon Pin3(38)
EXT_ENC_PRESS | MCU Pin46 PB0 (shared with beeper) | Ribbon Pin4(37)

I would like to retrofit this and enable it, but I don't see any place in the firmware to define the above pins. These are active-high contact closures to +3.3V. Can this be accommodated, and if not then may I know some hints about where in the source code to look? I am not a programmer but am willing to spend some time on it.

snovotill, 2021-07-17:
Ribbon cable pinout is in attached document. This is used for instance on TronXY-2Pro
[XY-2Pro-Titan CXY-V6-19107.zip](https://github.com/rhapsodyv/Marlin/files/6834606/XY-2Pro-Titan.CXY-V6-19107.zip)

nubbyless, 2021-08-03:
Very interesting anyplace i can follow your progress