rhapsodyv/Marlin #42: Newer tronxy x5sa does not work with your source not does the update.cbd in latest bugfix.

me2151, 2021-07-29:
### Description

Update.cbd nor direct flash of your source does not work with newer tronxy x5sa non pro tmc. I believe I have a bit of a rare/newer version. My board is a v6 with tmc, my boot version is 1.58 and software version is 1.19. according to octoprint(I am not using the CBD fix plugin) the software is actually marlin 2.0.7.2(April 17 build) from the factory. I am assuming the bootloader's header is updated so the update.cbd file does not update the firmware. Directly flashing the printer with firmware.bin using stm32core successful flashes but it does not boot at all( no screen, still shows as a com port in stm32core. Octoprint will not connect.). Using the latest bugfix with my own config I can boot marlin using direct flash but the update.cbd still does not work(tested with stock firmware on, not marlin).

### Steps to Reproduce

1. Select my printer using the config you provide
2. Select a screen mode (tft_color_ui)
3. Build
4a. Put update.cbd on sdcard, place sdcard in printer, power on printer.
4b. Connect printer to stm32core, erase all sectors, program firmware.bin(this step works fine on newest bugfix)

**Expected behavior:** sd update.cbd or firmware.bin flash to install marlin and boot.

**Actual behavior:** update.cbd boots printer as normal on stock firmware, direct flash does nothing.

#### Additional Information

* If you could provide steps to get the encryption info from the stock firmware I can update it for myself since I have a bit of a rare case. Or I can provide the firmware dump to you so you can check it out yourself.
* 
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

makmillion, 2021-11-16:
You have a newer Chitu board which is already running Marlin 2.0.7.2. You will need to follow [https://tronxy.fandom.com/wiki/Flashing_Marlin_on_2021_Tronxy_Boards](this guide) after confirming that your board is running Marlin. You can confirm this by connecting to Pronterface, or similar, by USB, and running M115.

me2151, 2021-11-17:
So no new info from my initial post. Even then the configuration provided doesn't work. Doesn't matter to me anyway now as I made my own that works. Also as stated in first post.

closed 2021-11-17 by me2151