rhapsodyv/Marlin #44: Question - Does the V6 board hardware support Bed PID setup, or will it cause overheating components?

Trellian, 2021-09-01:
Hello, does anyone know if the V6 board hardware support Bed PID setup, or will it cause overheating components?  I see this note in **configuration.h**:

***code
        /**
         * PID Bed Heating
         *
         * If this option is enabled set PID constants below.
         * If this option is disabled, bang-bang will be used and BED_LIMIT_SWITCHING will enable hysteresis.
         *
         * The PID frequency will be the same as the extruder PWM.
         * If PID_dT is the default, and correct for the hardware/configuration, that means 7.689Hz,
         * which is fine for driving a square wave into a resistive load and does not significantly
         * impact FET heating. This also works fine on a Fotek SSR-10DA Solid State Relay into a 250W
         * heater. If your configuration is significantly different than this and you don't understand
         * the issues involved, don't use bed PID until someone else verifies that your hardware works.
         */

lenigma1, 2021-09-01:
I too have a Tronxy with the stock Chitu V6 Mobo.
PID is all I use.
I'm thankful I was told what it was, and how to calibrate.
It does offer a more stable temperature setting.
No issues with PID to date, I'm happy to report.
I hope this helps!
Cheers😁

Trellian, 2021-09-01:
Thank you! Can you confirm that you are using PID on the **Bed**, and not just the Hot-end?

makmillion, 2021-09-01:
I have three (3) XY-2 Pro Titan with V6 boards, running Marlin from this fork, and all three are using PID Bed. They all work fine and are run almost constantly at 80c.

Trellian, 2021-09-02:
@makmillion @lenigma1  thank you both!

closed 2021-09-02 by Trellian

lenigma1, 2021-10-02:
> Thank you! Can you confirm that you are using PID on the **Bed**, and not just the Hot-end?

I'll go one step further, and state that although it was some trial and error to find the needed pins, before a upgraded to a Duet 3 mini 5+ last week(night and day better than ChiTu boards!), I had a perfectly working PID for Bed, HotEnd, AND chamber via a 3rd thermocouple(wanted to add 2, one on either side of chamber... but couldnt find a 4th pin for thermo's...) and a 500 watt, high accuracy PTC heater and a couple 120mm x 25 Noctua's.

Trellian, 2021-10-02:
@lenigma1 Wow, well done! That sounds like an excellent setup :)

And I got my bed PID working great, thank you!