rhapsodyv/Marlin #45: [Tronxy X5SA 400 Pro] update.cbd does not work

hexsmith, 2021-09-02:
Hi.  I'm trying to flash Marlin on my new Tronxy X5SA 400 Pro.  I have a CXY-V6-191017 board, and I am able to compile with Visual Studio Code and PlatformIO.

I have tried several things, including compiling for F103, with no success.

OEM firmware is utter garbage, and it constantly disconnects from OctoPrint.  Please advise.

Pedro8005x, 2021-09-02:
Hi,
There are solutions for that.
Registration may be required.
https://www.facebook.com/groups/169728600253557/permalink/963539657539110/

hexsmith ***@***.***> ezt írta (időpont: 2021. szept. 2., Csü
16:17):

hexsmith, 2021-09-05:
I'm not on Facebook, and I refuse to use it.  Please provide an alternative link.  Thanks.

Pedro8005x, 2021-09-05:
https://tronxy.fandom.com/wiki/Flashing_Marlin_on_2021_Tronxy_Boards

hexsmith ***@***.***> ezt írta (időpont: 2021. szept. 5., V
2:10):