rhapsodyv/Marlin #54: does not compile (anymore?)

Diiescher, 2023-07-12:
hi, 
I hope this is still with maintenance :)

I got my hands on a rather old Tronxy XY2 Pro with a V5 Mainboard and would like to try out Marlin on this thing.

Following this guide (https://github.com/EddyBeaupre/Tronxy-XY2-Pro#restore-stock-tronxy-firmware-and-settings) to the point I only get this result when building 
```
 *  Executing task in folder Marlin-tronxy-full-mks-tft2: C:\Users\Oli\.platformio\penv\Scripts\platformio.exe run 

Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead      
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Processing chitu_f103 (platform: ststm32@~12.1; board: marlin_CHITU_F103; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_CHITU_F103.html
PLATFORM: ST STM32 (12.1.1) > CHITU STM32F103Z (64k RAM. 512k Flash)
HARDWARE: STM32F103ZET6 72MHz, 59.12KB RAM, 469.03KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple @ 3.10000.201129 (1.0.0)
 - tool-stm32duino @ 1.0.2
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing SoftwareSerialM
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: SoftwareSerialM@1.0.0 has been installed!
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SoftwareSerialM @ 1.0.0
|-- STM32ADC @ 1.0
|-- EEPROM
|-- USBComposite for STM32F1 @ 0.99
|-- Wire @ 1.0
|-- Servo(STM32F1) @ 1.1.2
Building in release mode
RuntimeError: deque mutated during iteration:
  File "C:\Users\Oli\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 182:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "C:\Users\Oli\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Oli\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Oli\Desktop\Tronxy\Firmware\Marlin\Repository\Marlin-tronxy-full-mks-tft2\buildroot\share\PlatformIO\scripts\chitu_crypt.py", line 8:
    marlin.relocate_firmware("0x08008800")
  File "C:\Users\Oli\Desktop\Tronxy\Firmware\Marlin\Repository\Marlin-tronxy-full-mks-tft2\buildroot\share\PlatformIO\scripts\marlin.py", line 27:
    replace_define("VECT_TAB_ADDR", address)
  File "C:\Users\Oli\Desktop\Tronxy\Firmware\Marlin\Repository\Marlin-tronxy-full-mks-tft2\buildroot\share\PlatformIO\scripts\marlin.py", line 20:
    for define in env['CPPDEFINES']:
======================================================================================================================================= [FAILED] Took 8.98 seconds =======================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
chitu_f103     FAILED    00:00:08.977
================================================================================================================================== 1 failed, 0 succeeded in 00:00:08.977 ==================================================================================================================================
 *  The terminal process "C:\Users\Oli\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

branch: tronxy-full-mks-tft
Platform IO: Core 6.1.9·Home 3.4.4
VS Code: Version: 1.80.0 (user setup)
OS: Windows_NT x64 10.0.19045

RasForsberg, 2023-08-18:
I have the same problem trying to compile a build for my V8 board (Should be the same as a V5 board) I have tried to compile other options as well, but all fail.

Did you find a solution or are you still waiting?

Diiescher, 2023-08-20:
> Did you find a solution or are you still waiting?

still waiting...
It looks like this project has been abandoned and unfortunately I am not skilled enough to fix it.

clwebb89, 2023-10-05:
I have the same problem. I am not skilled enough to fix this. I have a X5SA and I can make a build with the example configs but it does not compile correctly. It still does give me an update. cbd file and works but seems sketchy. 

What needs to be done to fix this?

Pedro8005x, 2023-12-27:
> Solution is , edit marlin.py and change lines to:
>
> def replace_define(field, value):
> envdefs = env['CPPDEFINES'].copy()
> for define in envdefs:
> if define[0] == field:
> env['CPPDEFINES'].remove(define)
> env['CPPDEFINES'].append((field, value))

Great, thanks.

Otherwise, I had success with a non-latest official version of marlin also,
with Marlin-2.0.9.7.zip
<https://codeload.github.com/MarlinFirmware/Marlin/zip/refs/tags/2.0.9.7> ,
after some "winmerge" of config files. I had to wire UART pins to avoid
random motor stops and needed a new compile for changes. Motors are stable
since this mod, but linear advance enable is not tested yet, it was my
instan-failure flag with my v6 mainboard.

Márcio Carlos ***@***.***> ezt írta (időpont: 2023. dec. 27.,
Sze, 4:10):

kenci, 2023-12-29:
The real reason is that in Marlin/buildroot/share/PlatformIO/scripts/marlin.py the iterated environment is getting modified during the iteration in the replace_define() function, update it as follows and the error will be gone:

```
def replace_define(field, value):
    found_define = None
    for define in env['CPPDEFINES']:
        if define[0] == field:
            found_define = define
            break
    if found_define:
        env['CPPDEFINES'].remove(found_define)
    env['CPPDEFINES'].append((field, value))
```

CoderMal, 2024-04-18:
Anyone get this to compile?